clydebarrow/esphome #4: Use ESPHome font generator

clydebarrow, 2023-11-23:
The fonts created by the ESPHome font component are close to what LVGL needs, but don't currently work.

Some changes to the font generator will be needed to make this happen.

nagyrobi, 2024-01-16:
Just found this note at the bottom of the list... Adding my use case as a reminder.

The benefit would not only be usage of various fonts for the texts, but rather a very handy way to display custom symbols in the text fields - eg. icons on buttons.

For example this works with the standard renderer:

```yml
font:
  - file: 'custom/materialdesignicons-webfont.ttf'
    id: mdiicons
    size: 84
    glyphs: [
      "\U000f0084", # mdi-battery-charging
      "\U000f0091", # mdi-battery-unknown
      ]
  - file: 'custom/my_own_fontforge_crafted_fontfile.ttf'
    id: robicon_42
    size: 42
    glyphs: [
      "\u0670", #cover 100
      "\u0671", #cover 75
      "\u0672", #cover 50
      "\u0673", #cover 25
      "\u0674", #cover 0
      "\u0653", #ikea death star lamp
      ]

display:
  - platform: ...
    lambda: |-
      it.printf(10, 10, id(mdiicons), Color(0, 200, 0), TextAlign::LEFT, "\U000f0084");
      it.printf(10, 100, id(robicon_42), Color(100, 200, 50), TextAlign::LEFT, "\u0674");
      it.printf(10, 150, id(robicon_42), Color(100, 200, 50), TextAlign::LEFT, "\u0673");
      it.printf(10, 200, id(robicon_42), Color(100, 200, 50), TextAlign::LEFT, "\u0672");
      it.printf(10, 250, id(robicon_42), Color(100, 200, 50), TextAlign::LEFT, "\u0671");
```
Since icons are at unicode codepoints far from what's reachable on a standard keyboard, it's needed to specify the unicode codepoint of the glyph as a hex address escaped with `\u` or `\U`. This escape character probably needs to be handled at the widget's `text` configuration level.

Code points up to `0xFFFF` are encoded like `\uE6E8`. Lowercase `\u` and exactly 4 hexadecimal digits.
Code points above `0xFFFF` are encoded like `\U0001F5E9` . Capital `\U` and exactly 8 hexadecimal digits.

We could put icons this way on any widget supporting texts. And we could simply change them dynamically by updating the string containing the codepoint, changing color is nothing else than just update the color of the text.

nagyrobi, 2024-01-16:
With the same font config as above, this lvgl config:
```
        - label:
            x: 10
            y: 100
            id: lbl_id3
            text_font: robicon_42
            text: "aAűŰ"
            border_color: 0
            border_opa: 50%
            border_width: 1
        - label:
            x: 150
            y: 110
            id: lbl_id4
            text_font: robicon_42
            text: "\u0670"
            border_color: 0
            border_opa: 50%
            border_width: 1
```
Results in:

[image]

nagyrobi, 2024-02-01:
This now works nicely, thank you, except:

> Code points above `0xFFFF` are encoded like `\U0001F5E9` . Capital `\U` and exactly 8 hexadecimal digits.

Only code points up to `0xFFFF` work.
```yaml
font:
  - file: 'custom/RobotoCondensed-Regular.ttf'
    id: robi_20
    size: 20
    bpp: 4
    glyphs: [
      0,1,2,3,4,5,6,7,8,9,A,B,C,D,E,F,G,H,I,J,K,L,M,N,O,P,Q,R,S,T,U,V,W,X,Y,Z,a,b,c,d,e,f,g,h,i,j,k,l,m,n,o,p,q,r,s,t,u,v,w,x,y,z,á,Á,é,É,í,Í,ó,Ó,ö,Ö,ő,Ő,ú,Ú,ü,Ü,ű,Ű,
      (,),+,-,_,.,°,•,µ,
      "\u0020", #space
      "\u0021", #!
      "\u0022", #"
      "\u0025", #%
      "\u0027", #'
      "\u002C", #,
      "\u003A", #:
      "\u003D", #=
      "\u003F", #?
      ]
    extras:
      - file: 'custom/materialdesignicons-webfont.ttf'
        glyphs: [
          "\U000F0C5F", #close-network-outline
          "\U000f0084", # mdi-battery-charging
          "\U000f0091", # mdi-battery-unknown
          ]
```
and
```
    widgets:
      - label:
          text: "\U000f0084" #mdi-battery-charging
          text_font: robi_20
```
causes error:
```
[I][lvgl:000]: [Warn] (1094214.020, +908)      lv_draw_sw_letter: lv_draw_letter: glyph dsc. not found for U+F0084    (in lv_draw_sw_letter.c line #106)
```

nagyrobi, 2024-02-01:
Tried by converting the codepoint adress to a [surrogate pair](https://www.russellcottrell.com/greek/utilities/SurrogatePairCalculator.htm), where `\U000f0084` = `\uDB80\uDC84`.

For:
```yaml
    widgets:
      - label:
          text: "\uDB80\uDC84" #mdi-battery-charging
          text_font: robi_20
```
results in:
```
Traceback (most recent call last):
  File "/root/esphome/venv/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/esphome/esphome/__main__.py", line 1041, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/root/esphome/esphome/__main__.py", line 1028, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/esphome/esphome/__main__.py", line 407, in command_compile
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/root/esphome/esphome/__main__.py", line 192, in write_cpp
    generate_cpp_contents(config)
  File "/root/esphome/esphome/__main__.py", line 204, in generate_cpp_contents
    CORE.flush_tasks()
  File "/root/esphome/esphome/core/__init__.py", line 681, in flush_tasks
    self.event_loop.flush_tasks()
  File "/root/esphome/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "/root/esphome/esphome/__main__.py", line 184, in wrapped
    await coro(conf)
  File "/root/esphome/esphome/components/lvgl/__init__.py", line 1633, in to_code
    (_, ext_init) = await widget_to_code(
                    ^^^^^^^^^^^^^^^^^^^^^
  File "/root/esphome/esphome/components/lvgl/__init__.py", line 1803, in widget_to_code
    init.extend(await fun(obj, w_cnfig))
                ^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/esphome/esphome/components/lvgl/__init__.py", line 1020, in label_to_code
    init.append(f"lv_label_set_text({var}, {value})")
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/esphome/esphome/cpp_generator.py", line 255, in __str__
    return cpp_string_escape(self.string)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/esphome/esphome/helpers.py", line 58, in cpp_string_escape
    string = string.encode(encoding)
             ^^^^^^^^^^^^^^^^^^^^^^^
UnicodeEncodeError: 'utf-8' codec can't encode characters in position 0-1: surrogates not allowed
Compile failed
```
Also adding `"\uDB80\uDC84",` to the glyphs list of the font, results in a `Failed config` follwed by a screen full of characters...

nagyrobi, 2024-02-01:
In theory LVGL should support surrogate pairs, at least OpenHASP [uses them](https://www.openhasp.com/0.7.0/design/fonts/#1-as-json-payload) to display glyphs.

We should try to [calculate](https://stackoverflow.com/questions/8868432/how-are-surrogate-pairs-calculated) it during compile.

nagyrobi, 2024-02-01:
font validator: in yaml allow `otf` and `woff` file exensions alongsite `ttf`, because `pillow` supports them (renamed a `woff` to ttf, it worked).

clydebarrow, 2024-02-01:
> In theory LVGL should support surrogate pairs, at least OpenHASP [uses them](https://www.openhasp.com/0.7.0/design/fonts/#1-as-json-payload) to display glyphs.

No, surrogates are only part of UTF-16 which there is no particular need to support. UTF-8 works, as does using a full codepoint.

The issue with codepoints above 0xFFFF is now fixed.

closed 2024-02-01 by clydebarrow