clydebarrow/esphome #10: lvgl.obj.update - templatable state

nagyrobi, 2024-01-08:
Would simplify automations a bit if this would be possible:
```yml
    on_state:
      - lvgl.obj.update:
          id: btn_relay_1
          state:
            checked: !lambda "return x;"
```
 This option is not templatable.

Also it would be good if we had `on_state` for the LVGL switch/etc...

clydebarrow, 2024-01-08:
> it would be good if we had `on_state` for the LVGL switch/etc...

That needs `on_state` implemented in the core switch component, it's not an LVGL issue.

>           hidden: !lambda "return x;"

While it's possible to make this templatable, the variable `x` will still not be accessible, as this would only be available in a lambda directly under `on_state`. You can of course use `id(b_sensor).state` in any template.

nagyrobi, 2024-01-17:
OK, here's a different example:

```yaml
binary_sensor:
  - platform: homeassistant
    id: apa_lampa
    entity_id: light.apa_lampa
    on_state:
      then:
        if:
          condition:
            lambda: return x;
          then:
            - lvgl.widget.update:
                id: btn_apa_lampa
                state:
                  checked: true
          else:
            - lvgl.widget.update:
                id: btn_apa_lampa
                state:
                  checked: false
```
could be easier with:
```yaml
binary_sensor:
  - platform: homeassistant
    id: apa_lampa
    entity_id: light.apa_lampa
    on_state:
      lvgl.widget.update:
        id: btn_apa_lampa
        state:
          checked: !lambda return x;
```

clydebarrow, 2024-01-24:
this should be all good now.

closed 2024-01-24 by clydebarrow