clydebarrow/esphome #12: Extra widgets suggestion

nagyrobi, 2024-01-08:
Don't know if you plan to add all the widgets but I'd suggest these:
[ ] lv_animimg, compared to simple image only difference is that instead of one source image, you set an array of multiple source images. A flashing status icon, or battery charging for example.
[ ] lv_colorwheel, a color picker [ ] to set colors for lights.
[ ] lv_imgbtn, button showing different images based on state. That's for buttons showing eg. different lightbulb icons for on and off state
[ ] lv_keyboard, enter Wi-Fi credentials :smile:
[ ] lv_led, simple indicator for something ON or OFF
[ ] lv_menu, menus are popular in ESPHome, there's one for lcd displays so here is a must I guess
[ ] lv_spinbox, this is mandatory for thermostats
[ ] lv_spinner, I use this for my fan spinning speed visualisation, rotation speed can be customized

Not sure about
[ ] lv_msgbox
[ ] lv_win
These should be called with some actions...

nielsnl68, 2024-01-08:
The **lv_msgbox** is for sure interested to show notifications on the screen.
There is already a PR that added this functionality and can be used like:

```yaml
binary_sensor:
  - platform: gpio
    pin: GPIO0
    name: "Notificatiom Button"
    on_press:
      then:
       - lvgl.notification:
          title: "Hello World"
          message: "Please, brush your teeth every day."
          buttons: ["Done","Not Today"]
```
I hope @clydebarrow can add it soon.

clydebarrow, 2024-01-08:
Yeah, I saw that just earlier. I don't seem to have notifications on for PRs or issues :-( Will get to it as I work through the widgets.

nagyrobi, 2024-01-09:
How would a **lv_win** look like? Would it be easier than drawing the items manually? (a full screen translucent object to make it modal, children objects on top of it for interaction).

How do we get return states?

Use case:
[image]
Five windows with motorized shutters. In addition to the usual UP/STOP/DOWN buttons, the inhabitants want to have three shortcut positions for each shutter, and also a slider for free positioning. Useless to say that it's impossible to put all these on a page, and also it would be a pity to use up 5 pages just for these.

We'll just put the most used, UP/STOP/DOWN buttons on the page (this already looks a bit much...) and we'll use a long press event on the middle STOP button to show a pop-up with the extra required settings related to the desired shutter.

OpenHASP [solution](https://www.openhasp.com/0.7.0/integrations/home-assistant/sampl_conf/#shutter-control-panel)

nagyrobi, 2024-01-17:
You can create more complex dialogs added to `top_layer` with a full-screen sized, half-opaque ``obj`` with any child widgets on it, and the ``hidden`` flag set to ``true`` by default. For non-modal dialogs, simply set the ``clickable`` flag to ``false`` on it.

nielsnl68, 2024-01-17:
> Yeah, I saw that just earlier. I don't seem to have notifications on for PRs or issues :-( Will get to it as I work through the widgets.

still waiting to see it merged or commenting on.

nagyrobi, 2024-01-17:
> > Yeah, I saw that just earlier. I don't seem to have notifications on for PRs or issues :-( Will get to it as I work through the widgets.
> 
> still waiting to see it merged or commenting on.

Meanwhile it got implemented in a different way: https://deploy-preview-3510--esphome.netlify.app/components/lvgl#msgboxes

clydebarrow, 2024-01-17:
Yes I considered not implementing the inbuilt msgbox but decided it was nice to have a shorthand way to create dialogs.

nielsnl68, 2024-01-18:
> Meanwhile it got implemented in a different way: https://deploy-preview-3510--esphome.netlify.app/components/lvgl#msgboxes

I like my solution more sorry. I have already some use cases where i want to use this msgboxes. I do not see why we should  predefine them,