clydebarrow/esphome #9: Pagination implementation with `screen` object

nagyrobi, 2024-01-08:
`screens:` element

inspired from `select` concept

idea:
```
screen_loop: true #wrap arpound to the first/last when doing next/previous
screens:
  - screen:
    id: ...
    skip: false #skip this screen when doing next/previous (templatable)
    widgets:
      - ...
```

actions:
```
lvgl.screen.show
lvgl.screen.next
lvgl.screen.previous
```

nagyrobi, 2024-01-11:
Thanks for https://github.com/clydebarrow/esphome/commit/cec9698fc5d39f21df8aa8a89157bfe746c9867d!
Works!

There are use cases for selecting pages remotely, eg. display a page automatically when certain conditions are met in HA.
A simple way could be perhaps to automatically populate a `select` component with the pages.
This would require to support names for the pages (but only when `select` is configured, otherwise names can remain optional), and `select` should have a boolean to choose `include_skipped` if one wants to be able to remotely display skipped pages too.

clydebarrow, 2024-01-16:
Page wrap implemented (as default)

closed 2024-06-14 by nagyrobi