clydebarrow/esphome #20: Too many action types...?

nagyrobi, 2024-01-09:
I find it that there are too many actions registered to change some specific properties.

There's the powerful 
`lvgl.obj.update`
action to change properties/styles/states/flags of the widget main parts:
```
      - lvgl.obj.update:
          id: day_label
          bg_color: 0x00FF00
          state:
            disabled: false
```

And there are actions like
`lvgl.checkbox.update`
`lvgl.label.update`
`lvgl.arc.update`
`lvgl.roller.update`
to change properties/styles/states/flags of certain widget's subparts.
```yml
      - lvgl.label.update:
          id: day_label
          text: "Something"
```
Or even
`lvgl.indicator.line.update`
to change properties/styles/states/flags of certain widget's parts of the subparts.
```yml
      - lvgl.indicator.line.update:
          id: minute_hand
          value: 10
```
Are these intended to be like shorthands?

Wouldn't it make sense to structure everything under `lvgl.obj.update` (moreover, something simpler like `lvgl.update`) and nest the subparts hierarchically?

like:
```
      - lvgl.update:
          id: minute_hand
          line_color: 0x00FF00
          state:
            disabled: false
          indicator:  #could only be specifict to what the widget supports, eg. knob etc.
            value: 10
```
add
```
      - lvgl.update:
          id: day_label
          bg_color: 0x00FF00
          state:
            disabled: false
          text: "Something"  #additional widget-specific options just added at the same level as the main ones
```

nagyrobi, 2024-01-12:
I'd find useful a `lvgl.widget.delete` action (`lv_obj_del_async(obj)` or `lv_obj_del`).
Use case is a bootlogo, which could be fullscreen `obj` with a centered image, however just hiding it just keeps it uselessly in memory. We could free that resource...

clydebarrow, 2024-01-13:
The problem with consolidating all the actions under one is that some things, like `text:` are only applicable to certain types of widgets. By having a different action for `label` the schema for that action can include `text:` as an option, whereas the schema for `line:` would not `points` but not `text`.  At the time the schema is parsed the type of the object referenced by `id` may not be known, so checking that only the appropriate options are used gets complicated, whereas right now it will throw a clear error message if an inappropriate option is used.

clydebarrow, 2024-01-13:
> I'd find useful a `lvgl.widget.delete` action (`lv_obj_del_async(obj)` or `lv_obj_del`). Use case is a bootlogo, which could be fullscreen `obj` with a centered image, however just hiding it just keeps it uselessly in memory. We could free that resource...

The image data is stored in flash memory, so deleting the `img` object won't free much memory. If and when run-time loading of images e.g from urls is supported then memory management can be implemented as part of it, but right now there's very little to be gained.

nagyrobi, 2024-01-13:
I see thanks.

closed 2024-01-13 by nagyrobi