clydebarrow/esphome #22: Make `color` configurations coherent

nagyrobi, 2024-01-09:
In ESPHome's `color` config, we can specify colors with `hex: FF0000`.
Here we can also specify colors in hexa, like `border_color: 0x2F8CD8`. 

Would be nice if it wouldn't be required to put that `0x` in front of the RGB color code.

clydebarrow, 2024-01-09:
So you're suggesting to allow either `color: 0xFF0000` or `hex: FF0000` as synonyms? That is doable, but would require significant extra code (needs to be done everywhere color is used), and could be confusing in my opinion, for a small gain. You can define ESPHome colors and use the ID in the lvgl config already.

nagyrobi, 2024-01-10:
I'd just say to allow for example `border_color: 2F8CD8` instead of `border_color: 0x2F8CD8`.

clydebarrow, 2024-01-13:
it is difficult, because in YAML `2F8CD8` is a string, and while that could be then converted as a hex number during validation, something like `606060` would be parsed as a decimal number by the YAML parser before it's even seen by a validator, so it would have to be written as `"606060"` which defeats the purpose. And since a color ID can also be used, how do you tell if `fabcde` is a hex number  or an ID? So I think it's much better to have one consistent and unambigous way of specifying a color rather than shorthands that only work some of the time.

Allowing `#606060` would be nice, but # is a comment indicator in YAML so that won't work.

nagyrobi, 2024-01-13:
Ok, forget this.

closed 2024-01-13 by nagyrobi