clydebarrow/esphome #25: `img` options

nagyrobi, 2024-01-10:
From https://docs.lvgl.io/8.3/widgets/core/img.html some options:
- Label as an image
- Transparency (correlation with ESPHome's image transparency possibilities?)
- Recolor (`img_recolor_opa` adn `img_recolor` similar to `bg_*`)
- Mosaic
- Offset
- Rounded image
- Transformations (if not already done)

nagyrobi, 2024-01-12:
Supported encodings?

nagyrobi, 2024-01-16:
Some tests:

```yaml
        - obj:  # background for test transparency
            id: color_matte
            x: 0
            y: 40
            width: 240
            height: 240
            bg_color: 0x00ffff
            bg_opa: cover
            radius: 0
            pad_all: 0
            border_width: 0
            widgets:
              - img:
                  align: center
                  src: esphome_logo_transparent
                  id: img_transp
```
with:

```yaml
image:
  - file: https://esphome.io/_static/made-for-esphome-black-on-transparent.png
    id: esphome_logo_transparent
    resize: 220x72
    type: # as below
    use_transparency: # as below
```
with `type: TRANSPARENT_BINARY` (transparency enabled by default) results in:
[image]

with `type: GRAYSCALE` results in (same result with `dither: FLOYDSTEINBERG`):
[image]

with: `type: GRAYSCALE` and `use_transparency: true` results in (nothing):
[image]

with `type: RGB565` results in:
[image]

with `type: RGB565` and `use_transparency: true` results in:
[image]

with: `type: RGB24` results in:
[image]

with: `type: RGB24` and `use_transparency: true` results in:
[image]

with: `type: RGBA` (transparency enabled by default) results in:
[image]

nagyrobi, 2024-01-16:
Repeated some of the tests with SVG image instead of PNG:

```yaml
image:
  - file: https://esphome.io/_static/made-for-esphome-black-on-transparent.svg
    id: esphome_logo_transparent
    resize: 220x72
    type: # as below
    use_transparency: # as below
```
with `type: TRANSPARENT_BINARY` (transparency enabled by default) results in:
[image]

with: `type: GRAYSCALE` and `use_transparency: true` results in (nothing):
[image]

with `type: RGB565` and `use_transparency: true` results in:
[image]

nagyrobi, 2024-01-16:
I think situation is similar to the fonts, LVGL seems to use a proprietary format, and they have online and offline converter tools for images to C arrays: https://lvgl.io/tools/imageconverter

It's two approaches here: 
- the command line tool can be used before compile
- some other implementation needs to be done at runtime for https://github.com/clydebarrow/esphome/issues/14

clydebarrow, 2024-02-05:
Transparency now works, e.g.
```
image:
  - file: image.png
    resize: 200x200
    type: RGB565
    use_transparency: true
    id: image_png
```

clydebarrow, 2024-02-05:
`img_recolor` and friends added.

closed 2024-02-05 by clydebarrow

nagyrobi, 2024-02-05:
> Transparency now works

Hooray:
[image]