clydebarrow/esphome #26: `label` options

nagyrobi, 2024-01-10:
From https://docs.lvgl.io/8.3/widgets/core/label.html:
- Long modes options
- Text recolor
- Custom scrolling animations
- Symbols

nagyrobi, 2024-01-10:
Also new line with `\n`, and recoloring with `Write a #ff0000 red# word` don't seem to work. Would be good to also keep these modes because it's rather easy to generate such strings from HA templates or within ESPHome lambdas.

4ib3r, 2024-01-11:
Hello, I created PR with improvements (recolor and long modes) from this issue :wink:

clydebarrow, 2024-01-15:
> Until recently it was possible to start with an empty string `text: ""`. Now this doesn't work

Fixed.

The animation stuff should be a separate issue to implement custom animations generally.

closed 2024-01-15 by clydebarrow