clydebarrow/esphome #34: Event triggers

nagyrobi, 2024-01-11:
A thought on events from:
https://docs.lvgl.io/8.3/overview/event.html

A universal `on_event` trigger could be implemented with a parameter `type` which can be chosen from an enum of LVGL events.
A few shorthands could be implemented for most popular events:

`on_press`-> LV_EVENT_PRESSED 
`on_release`-> LV_EVENT_RELEASED 
`on_click`-> LV_EVENT_SHORT_CLICKED 
`on_long_press`-> LV_EVENT_LONG_PRESSED 
`on_value`-> LV_EVENT_VALUE_CHANGED, returning the value in x (slider has been moved to value x, switch/btn toggled)

nagyrobi, 2024-01-11:
Fyi OpenHASP choose these: https://www.openhasp.com/0.7.0/design/objects/#events

nagyrobi, 2024-01-12:
We need `on_value` trigger because `btnmatrix` control flag `CLICK_TRIG` offers us this in a configurable way.

nagyrobi, 2024-01-12:
Also something like `on_draw_end` (perhaps `LV_EVENT_DRAW_POST_END`). Use case: after displaying a bootlogo on the top_layer, start a delay of a few seconds, and [ditch](https://github.com/clydebarrow/esphome/issues/20#issuecomment-1889806498) the logo widget...

clydebarrow, 2024-01-16:
On gesture and on_draw_end to be done at a later date maybe.

nagyrobi, 2024-01-17:
Use case:

```
        - btn:
            checkable: true
            id: btn_apa_lampa
#            state:
#              checked: !lambda |-
#                return id(apa_lampa).state;
            widgets:
              - label:
                  align: center
                  text_color: 0xFFFFFF
                  text: 'Apa Lampa'
            on_value:
              then:
                if:
                  condition:
                    lambda: return x;
                  then:
                    - homeassistant.service:
                        service: light.turn_on
                        data: 
                          entity_id: light.apa_lampa
                  else:
                    - homeassistant.service:
                        service: light.turn_off
                        data: 
                          entity_id: light.apa_lampa
```