clydebarrow/esphome #38: `one_checked` option for `btnmatrix`

nagyrobi, 2024-01-12:
From: https://docs.lvgl.io/8.3/widgets/core/btnmatrix.html#one-check

It's for making possible to check only one button from the matrix. When you toggle a button, the others will be "de-toggled". 
It would also be nice to complement this feature with the `select` functionality.

clydebarrow, 2024-01-13:
> `one_checked` option from: https://docs.lvgl.io/8.3/widgets/core/btnmatrix.html#one-check

Already implemented.

> we need on_value trigger to retrieve in x the ID of the button that was pressed.

Why? Each button in a button matrix can have its own actions. Although internally they are different to regular buttons, this is masked via the yaml config.

nagyrobi, 2024-01-13:
> Why? Each button in a button matrix can have its own actions. Although internally they are different to regular buttons, this is masked via the yaml config.

For the users to easier implement something like [key_collector](https://esphome.io/components/key_collector.html) to design an alarm input panel for example.

It's like when you need to input a sequence of numbers like a pin code - without having to define complicated scripts to concatenate strings located in global variables.
Actually, `key_collector` is a pretty well thought out component for this purpose. 

Also see https://github.com/clydebarrow/esphome/issues/34#issuecomment-1889288429

nagyrobi, 2024-01-17:
> > `one_checked` option from: https://docs.lvgl.io/8.3/widgets/core/btnmatrix.html#one-check
> 
> Already implemented.

Tested, config passes through the schema, but doesn't have any effect. Multiple buttons can stay checked.

clydebarrow, 2024-01-17:
> Actually, `key_collector` is a pretty well thought out component for this purpose. Maybe we could make `btnmatrix` work with it?

Or maybe make `key_collector` work with LVGL.

clydebarrow, 2024-01-17:
> ested, config passes through the schema, but doesn't have any effect. Multiple buttons can stay checked.

Fixed.

closed 2024-01-17 by clydebarrow

ssieb, 2024-01-17:
If the `btnmatrix` provides the `key_provider` interface, then it will work automatically.

nagyrobi, 2024-01-17:
@ssieb Can you show a sample?

ssieb, 2024-01-17:
Check out the `wiegand` and `matrix_keypad` components for example.  It's very simple.

nagyrobi, 2024-01-17:
@clydebarrow maybe we could add a config option `key_provider: true` to btnmatrix root, and simply provide the texts put on the buttons to it as keys (and not allow symbols in this case).

clydebarrow, 2024-01-17:
That will work, no need for the config option, and it will just send the first character of the button label (key_provider can only send single characters.)

ssieb, 2024-01-17:
Symbols are fine.  If you want to send more than a single character, then just call the send method multiple times.

nagyrobi, 2024-01-18:
@ssieb symbols here are graphical elements, not just *# keys.

@clydebarrow I thought about confic option because one may use multiple btnmatrix instances (eg page navigation etc) and likely wouldn't want to use key_provider with those.

clydebarrow, 2024-01-18:
Symbols in the LVGL world are just Unicode strings.

The btnmatrix won't send any keys unless it's connected to a key_collector.

ssieb, 2024-01-18:
The key interface only handles `uint8_t` right now because it originally came from the keypad component.  It could be expanded, but I'm not sure how useful that would be.  What would you be using the keys for?
This should probably be a discord discussion. :-)

nagyrobi, 2024-01-18:
> The btnmatrix won't send any keys unless it's connected to a key_collector.

Oh, you'd put at `source_id` the btnmatrix id?

ssieb, 2024-01-18:
Yes.  As with most things in esphome, if there's only one source, it will pick it up automatically, but otherwise, you need to specify the source id.

clydebarrow, 2024-01-19:
> If the `btnmatrix` provides the `key_provider` interface, then it will work automatically.

Done. Works a treat.

nagyrobi, 2024-01-26:
Please reopen this, I'm trying to print the keys to a label but it won't compile:
```yaml
key_collector:
  - id: pincode_reader
    source_id: lvgl_keypad
    min_length: 4
    max_length: 4
    end_keys: "#"
    end_key_required: true
    back_keys: "*"
    allowed_keys: "0123456789"
    timeout: 5s
    on_progress:
      - logger.log:
          format: "input progress: '%s', started by '%c'"
          args: [ 'x.c_str()', "(start == 0 ? '~' : start)" ]
      - lvgl.label.update:
          id: keypad_label
          text: !lambda 'return str_sprintf("%s", x.c_str());'
          border_color: 0
    on_result:
      - logger.log:
          format: "input result: '%s', started by '%c', ended by '%c'"
          args: [ 'x.c_str()', "(start == 0 ? '~' : start)", "(end == 0 ? '~' : end)" ]
      - lvgl.label.update:
          id: keypad_label
          text: !lambda 'return str_sprintf("%s", x.c_str());'
          border_color: 0xFF0000
    on_timeout:
      - logger.log:
          format: "input timeout: '%s', started by '%c'"
          args: [ 'x.c_str()', "(start == 0 ? '~' : start)" ]
      - lvgl.label.update:
          id: keypad_label
          text: !lambda 'return str_sprintf("%s", x.c_str());'
          border_color: 0
```
fails to compile:
```
<unicode string>: In lambda function:
<unicode string>:1612:34: error: 'x' was not declared in this scope
<unicode string>: In lambda function:
<unicode string>:1620:34: error: 'x' was not declared in this scope
<unicode string>: In lambda function:
<unicode string>:1628:34: error: 'x' was not declared in this scope
```
It seems the result of the keycollector is available in the logger action but not the LVGL action.

nagyrobi, 2024-01-27:
Recipe ready for the Cookbook: https://deploy-preview-3510--esphome.netlify.app/cookbook/lvgl#a-numeric-input-keypad