clydebarrow/esphome #40: Escape characters in texts, eg. `label`

nagyrobi, 2024-01-16:
`label` supports multiline texts: https://docs.lvgl.io/8.3/widgets/core/label.html#newline
`\n` should start a new line. Currently:

```
        - label:
            align: center
            id: lbl_id
            y: 30
            recolor: true
            text: 'Write a #ff0000 red# word\nand a #0000ff blue# one'
```
produces:  
[image]
looks like `\n` is interpreted textually, not being escaped.

Other escape sequences like `\u`, `\U` [should be supported](https://github.com/clydebarrow/esphome/issues/4#issuecomment-1893295713) on long term...

clydebarrow, 2024-02-02:
For escape sequences like newline to be translated, enclose the string in double quotes: 
```
"Write a #ff0000 red# word\nand a #0000ff blue# one"

closed 2024-02-02 by clydebarrow

nagyrobi, 2024-02-03:
😬 can't believe I missed that, sorry.