clydebarrow/esphome #43: `state` options templatable

nagyrobi, 2024-01-17:
To implement like this:
```
        - btn:
            checkable: true
            id: button_light
            state:
              checked: !lambda |-
                return id(ha_light_binary_sensor).state;
```
instead of tedious `on_state`/`on_click` automations at the binary sensor...

nielsnl68, 2024-01-17:
That would make my live much easier :D

nagyrobi, 2024-01-17:
Yeah also stuff like:
```
            state:
              disabled: !lambda |-
                return !id(ha_available).state;
```
To prevent interactions with widgets related to remote entities through HA, if disconnected.

nagyrobi, 2024-01-17:
Request applies also to the `.update` action https://github.com/clydebarrow/esphome/issues/10#issuecomment-1895594838

closed 2024-01-17 by clydebarrow