clydebarrow/esphome #47: Outside alignment not working

nagyrobi, 2024-01-23:
```yaml
        - img:
            x: 10
            y: 170
            src: cat_image
            id: img_id
            radius: 11
            clip_corner: true
            widgets:
              - label:
                  id: lbl_cathome
                  align: OUT_BOTTOM_MID
                  text: 'Home'
                  text_color: 0xFF0000
```
results in
[image]

Expectation is to have the text under the picture because of `align: OUT_BOTTOM_MID`.
Ref: https://docs.lvgl.io/8/overview/coords.html#align

clydebarrow, 2024-01-23:
A child widget is clipped to its parent boundaries anyway, so setting an OUT alignment will never work (and is in fact ignored.)

I've updated the schema to disallow OUT values on `align:` and added `align_to:` to allow a widget to be aligned to another.
The two objects should be siblings.

```
        - img:
            x: 10
            y: 200
            src: cat_image
            id: img_id
            radius: 11
            clip_corner: true
        - label:
            id: lbl_cathome
            align_to:
              id: img_id
              x: 0
              y: 5
              align: OUT_BOTTOM_MID
            text: 'Home'
            text_color: 0x00FF00
```
The x and y are optional and default to 0.

closed 2024-01-23 by clydebarrow

nagyrobi, 2024-01-24:
This is more powerful than I thought. Makes alignment of the widgets a whole lot easier.

nagyrobi, 2024-01-24:
```
<unicode string>:1867:45: error: ‘LV_ALIGN_OUT_CENTER’ was not declared in this scope; did you mean ‘LV_ALIGN_CENTER’?
*** [.pioenvs/esp32-test-lvgl-host/src/main.o] Error 1
```

Of course, because `OUT_CENTER` indeed doesn't make sense. To be removed from https://github.com/clydebarrow/esphome/blob/750b432dd77c91efeecdcbfdafd880d11d554d33/esphome/components/lvgl/defines.py#L146