clydebarrow/esphome #48: Built-in fonts

nagyrobi, 2024-01-24:
```
Compiling .pioenvs/esp32-test-lvgl-host/src/main.o
<unicode string>: In function ‘void setup()’:
<unicode string>:2204:62: error: ‘montserrat_8’ was not declared in this scope
*** [.pioenvs/esp32-test-lvgl-host/src/main.o] Error 1
```
and 
```
Compiling .pioenvs/esp32-test-lvgl-host/src/main.o
<unicode string>: In function ‘void setup()’:
<unicode string>:2204:62: error: ‘montserrat_10’ was not declared in this scope
*** [.pioenvs/esp32-test-lvgl-host/src/main.o] Error 1
```
and
```
Compiling .pioenvs/esp32-test-lvgl-host/src/main.o
<unicode string>: In function ‘void setup()’:
<unicode string>:2204:62: error: ‘simsun_16_cjk’ was not declared in this scope
*** [.pioenvs/esp32-test-lvgl-host/src/main.o] Error 1
```

although they are compiled:
```
Compiling .pioenvs/esp32-test-lvgl-host/lib09a/lvgl/font/lv_font.o
Compiling .pioenvs/esp32-test-lvgl-host/lib09a/lvgl/font/lv_font_dejavu_16_persian_hebrew.o
Compiling .pioenvs/esp32-test-lvgl-host/lib09a/lvgl/font/lv_font_fmt_txt.o
Compiling .pioenvs/esp32-test-lvgl-host/lib09a/lvgl/font/lv_font_loader.o
Compiling .pioenvs/esp32-test-lvgl-host/lib09a/lvgl/font/lv_font_montserrat_10.o
Compiling .pioenvs/esp32-test-lvgl-host/lib09a/lvgl/font/lv_font_montserrat_12.o
Compiling .pioenvs/esp32-test-lvgl-host/lib09a/lvgl/font/lv_font_montserrat_12_subpx.o
Compiling .pioenvs/esp32-test-lvgl-host/lib09a/lvgl/font/lv_font_montserrat_14.o
Compiling .pioenvs/esp32-test-lvgl-host/lib09a/lvgl/font/lv_font_montserrat_16.o
Compiling .pioenvs/esp32-test-lvgl-host/lib09a/lvgl/font/lv_font_montserrat_18.o
Compiling .pioenvs/esp32-test-lvgl-host/lib09a/lvgl/font/lv_font_montserrat_20.o
Compiling .pioenvs/esp32-test-lvgl-host/lib09a/lvgl/font/lv_font_montserrat_22.o
Compiling .pioenvs/esp32-test-lvgl-host/lib09a/lvgl/font/lv_font_montserrat_24.o
Compiling .pioenvs/esp32-test-lvgl-host/lib09a/lvgl/font/lv_font_montserrat_26.o
Compiling .pioenvs/esp32-test-lvgl-host/lib09a/lvgl/font/lv_font_montserrat_28.o
Compiling .pioenvs/esp32-test-lvgl-host/lib09a/lvgl/font/lv_font_montserrat_28_compressed.o
Compiling .pioenvs/esp32-test-lvgl-host/lib09a/lvgl/font/lv_font_montserrat_30.o
Compiling .pioenvs/esp32-test-lvgl-host/lib09a/lvgl/font/lv_font_montserrat_32.o
Compiling .pioenvs/esp32-test-lvgl-host/lib09a/lvgl/font/lv_font_montserrat_34.o
Compiling .pioenvs/esp32-test-lvgl-host/lib09a/lvgl/font/lv_font_montserrat_36.o
Compiling .pioenvs/esp32-test-lvgl-host/lib09a/lvgl/font/lv_font_montserrat_38.o
Compiling .pioenvs/esp32-test-lvgl-host/lib09a/lvgl/font/lv_font_montserrat_40.o
Compiling .pioenvs/esp32-test-lvgl-host/lib09a/lvgl/font/lv_font_montserrat_42.o
Compiling .pioenvs/esp32-test-lvgl-host/lib09a/lvgl/font/lv_font_montserrat_44.o
Compiling .pioenvs/esp32-test-lvgl-host/lib09a/lvgl/font/lv_font_montserrat_46.o
Compiling .pioenvs/esp32-test-lvgl-host/lib09a/lvgl/font/lv_font_montserrat_48.o
Compiling .pioenvs/esp32-test-lvgl-host/lib09a/lvgl/font/lv_font_montserrat_8.o
Compiling .pioenvs/esp32-test-lvgl-host/lib09a/lvgl/font/lv_font_simsun_16_cjk.o
Compiling .pioenvs/esp32-test-lvgl-host/lib09a/lvgl/font/lv_font_unscii_16.o
Compiling .pioenvs/esp32-test-lvgl-host/lib09a/lvgl/font/lv_font_unscii_8.o
```
Not sure what `fmt_txt` and `loader` fonts are, but they are compiled, would probably worth to try and see.

---
with `text_font: montserrat_28_compressed` labels are empty, and the LVGL logs this:

```
[I][lvgl:000]: [Warn]   (397413.594, +0)         lv_font_get_bitmap_fmt_txt: Compressed fonts is used but LV_USE_FONT_COMPRESSED is not enabled in lv_conf.h  (in lv_font_fmt_txt.c line #131)
[I][lvgl:000]: [Warn]   (397413.594, +0)         lv_draw_sw_letter: lv_draw_letter: character's bitmap not found        (in lv_draw_sw_letter.c line #145)
```

with `text_font: montserrat_12_subpx` labels are empty, and the LVGL logs this:
```
[I][lvgl:000]: [Warn]   (397578.880, +0)         lv_draw_sw_letter: Can't draw sub-pixel rendered letter because LV_USE_FONT_SUBPX == 0 in lv_conf.h    (in lv_draw_sw_letter.c line #153)
```

nagyrobi, 2024-01-24:
https://docs.lvgl.io/8/overview/font.html#subpixel-rendering this looks as something pretty hardware-specific:

> Subpixel rendering works only if the color channels of the pixels have a horizontal layout. That is the R, G, B channels are next each other and not above each other. 

You can't rotate the screen because of this.

We could ditch `montserrat_12_subpx`, also because this subpixel font is only available in12px size. It also requires special parameters at compile time. Imho if anyone wants to play with subpixel rendering in ESPHome, should try on a dedicated branch with custom fonts rendered.

clydebarrow, 2024-02-02:
Done

closed 2024-02-02 by clydebarrow