clydebarrow/esphome #49: `on_value` for more widgets

nagyrobi, 2024-01-24:
We need `on_value` for widgets supporting `checkable: true`, (`btn`, `checkbox`, `switch` etc) to make automations easier when the `LV_EVENT_VALUE_CHANGED` happens.

For examle here we have a checkable button with a label on it, and we want to show a differently coloured symbol based on the button's checked state.

If  `on_value` could return a binary value `x` in `checked` state, this would be easy to accomplish in the same yaml section, without having to rely on binary sensors etc.

```yml
        - btn:
            checkable: true
            id: btn_toggle_swicth
            widgets:
              - label:
                  id: lbl_btn_apa_lampa
                  align: center
                  text_color: 0xFFFFFF
                  symbol: CLOSE
            on_value:
              then:
                if:
                  condition:
                    lambda: return x;
                  then:
                    - lvgl.label.update:
                        id: lbl_btn
                        symbol: OK
                        text_color: 0xFFFF00
                    - homeassistant.service:
                        service: light.turn_on
                        data: 
                          entity_id: light.to_be_toggled
                  else:
                    - lvgl.label.update:
                        id: lbl_btn
                        symbol: CLOSE
                        text_color: 0xFFFFFF
                    - homeassistant.service:
                        service: light.turn_off
                        data: 
                          entity_id: light.to_be_toggled
```
---
 `dropdown` `roller` and `btnmatrix` could benefit from `on_value` if it would return the index `x` and the name `n` of the selected option after the `LV_EVENT_VALUE_CHANGED` occured.

```
        - dropdown:
            x: 80
            y: 180
            width: 105
            id: dropdown_id
            options:
              - Dry food
              - Wet food
              - Semi-moist
            on_value:
              - logger.log:
                  format: "Value index is: %.1f, name is %s"
                  args: [ 'x', 'n.c_str()' ]
              - homeassistant.service:
                  service: select.select_option
                  data:
                    entity_id: select.cat_feeder
                    option: !lambda return (n.c_str());
```
We need the name in a variable `x` as string because HA's `select.select_option` only offers selecting options by name, not index.

nielsnl68, 2024-01-24:
I was  just thinking when i was playing with the button on_click event.
Would it be nice to have the "homeassistant.service" build into the widgets directly.
so you have for example:

```yaml

     - btn:
            checkable: true
            id: btn_toggle_swicth
            widgets:
              - label:
                  id: lbl_btn_apa_lampa
                  align: center
                  text_color: 0xFFFFFF
                  symbol: CLOSE
            homeassistant.service:
               entity_id: light.to_be_toggled
               send: light.turn_off
               receive:
                  state:
                      checked: !lambda return x;

        - dropdown:
            x: 80
            y: 180
            width: 105
            id: dropdown_id
            options:
              - Dry food
              - Wet food
              - Semi-moist
            homeassistant.service:
               entity_id: light.to_be_toggled
               send: 
                  action: light.turn_off
                  option: !lambda return (n.c_str());
               receive:
                  value: !lambda return x;
```

nagyrobi, 2024-01-24:
`receive`?

nielsnl68, 2024-01-24:
Yes that would be like short way of:
```yaml
binary_sensor:
- platform: homeassistant
  publish_initial_state: true
  id: "light_totom"
  entity_id: light.woonkamer_kleuren_totom_light_1
  on_state:
    - script.execute: backlight_on    
    - lvgl.widget.update:
        id: lv_button0
        state:
          checked: !lambda return x;
```

nagyrobi, 2024-01-24:
But that's exclusive to `homeassistant.service`, right? Still need `on_value` for regular esphome automations...

nielsnl68, 2024-01-24:
true. for other sensors atc the `on_value` can be useful indeed.

nagyrobi, 2024-01-25:
Was thinking about this `homeassistant.service` and I think we still need the binary sensor though, to get the initial state of the entity from HA, to display it before calling the service.

nagyrobi, 2024-02-15:
Ran into a use case with a switch:

```yaml
              - switch:
                  id: swi_klima
                  x: 45
                  y: 68
                  align: TOP_MID
                  width: 70 #75
                  height: 40
                  on_release:
                    - if:
                        condition:
                          lambda: return lv_obj_get_state(id(swi_klima)) & LV_STATE_CHECKED;
                        then:
                          - homeassistant.service:
                              service: climate.set_hvac_mode
                              data:
                                hvac_mode: !lambda |-
                                  if (id(bs_climate_heat_mode).state == true) {
                                      return "heat";
                                  } else {
                                      return "cool";
                                  }
                                entity_id: climate.haloszoba_klima
                        else:
                          - homeassistant.service:
                              service: climate.turn_off
                              data:
                                entity_id: climate.haloszoba_klima
```

We could just use `x` here instead of `lv_obj_get_state(id(swi_klima)) & LV_STATE_CHECKED`.

Not sure if things like `on_turn_on` and `on_turn_off` or more LVGL-specialised `on_check` and `on_uncheck` would be worth it.

nagyrobi, 2024-06-14:
Works with all event triggers now.

closed 2024-06-14 by nagyrobi