clydebarrow/esphome #55: `lvgl.widget.update` for `btnmatrix`

nagyrobi, 2024-01-30:
butmat_szell is a btnmatrix widget.
I'd like to apply disabled state styles for the entire matrix:

`          - lvgl.widget.disable: buttons_szell`
or
```yaml
           - lvgl.widget.update:
               id: butmat_szell
               state:
                 disabled: true
```
causes error:
```
config/_02_host_lvgl_5_demos.yaml: In lambda function:
config/_02_host_lvgl_5_demos.yaml:105:26: error: cannot convert ‘esphome::lvgl::LvBtnmatrixType*’ to ‘lv_obj_t*’ {aka ‘_lv_obj_t*’}
In file included from .piolibdeps/esp32-test-lvgl-host/lvgl/lvgl.h:35,
                 from src/esphome/components/lvgl/lvgl_esphome.h:31,
                 from src/esphome.h:54,
                 from src/main.cpp:3:
.piolibdeps/esp32-test-lvgl-host/lvgl/src/core/lv_obj.h:261:36: note:   initializing argument 1 of ‘void lv_obj_clear_state(lv_obj_t*, lv_state_t)’
  261 | void lv_obj_clear_state(lv_obj_t * obj, lv_state_t state);
      |                         ~~~~~~~~~~~^~~
config/_02_host_lvgl_5_demos.yaml: In lambda function:
config/_02_host_lvgl_5_demos.yaml:142:24: error: cannot convert ‘esphome::lvgl::LvBtnmatrixType*’ to ‘lv_obj_t*’ {aka ‘_lv_obj_t*’}
.piolibdeps/esp32-test-lvgl-host/lvgl/src/core/lv_obj.h:253:34: note:   initializing argument 1 of ‘void lv_obj_add_state(lv_obj_t*, lv_state_t)’
  253 | void lv_obj_add_state(lv_obj_t * obj, lv_state_t state);
      |                       ~~~~~~~~~~~^~~
*** [.pioenvs/esp32-test-lvgl-host/src/main.o] Error 1
```

Having

```yaml
               state:
                 disabled: true
```

in btnmatrix's config doesnt have any effect.

clydebarrow, 2024-02-02:
`lvgl.widget.update` now works for button matrix.

Just set disabled state on the button matrix.

closed 2024-02-02 by clydebarrow

nagyrobi, 2024-02-05:
`lvgl.widget.enable` / `lvgl.widget.disable` cause this:

```
config/_02_host_lvgl_6_robi_migrate.yaml: In lambda function:
config/_02_host_lvgl_6_robi_migrate.yaml:2703:26: error: cannot convert ‘esphome::lvgl::LvBtnmatrixType*’ to ‘lv_obj_t*’ {aka ‘_lv_obj_t*’}
In file included from .piolibdeps/esp32-migr-lvgl-host/lvgl/lvgl.h:35,
                 from src/esphome/components/lvgl/lvgl_esphome.h:31,
                 from src/esphome.h:36,
                 from src/main.cpp:3:
.piolibdeps/esp32-migr-lvgl-host/lvgl/src/core/lv_obj.h:261:36: note:   initializing argument 1 of ‘void lv_obj_clear_state(lv_obj_t*, lv_state_t)’
  261 | void lv_obj_clear_state(lv_obj_t * obj, lv_state_t state);
      |                         ~~~~~~~~~~~^~~
config/_02_host_lvgl_6_robi_migrate.yaml: In lambda function:
config/_02_host_lvgl_6_robi_migrate.yaml:115:24: error: cannot convert ‘esphome::lvgl::LvBtnmatrixType*’ to ‘lv_obj_t*’ {aka ‘_lv_obj_t*’}
.piolibdeps/esp32-migr-lvgl-host/lvgl/src/core/lv_obj.h:253:34: note:   initializing argument 1 of ‘void lv_obj_add_state(lv_obj_t*, lv_state_t)’
  253 | void lv_obj_add_state(lv_obj_t * obj, lv_state_t state);
      |                       ~~~~~~~~~~~^~~
*** [.pioenvs/esp32-migr-lvgl-host/src/main.o] Error 1

```
and
```yaml
          - lvgl.widget.update:
              id: butmat_szell
              state:
                disabled: false # or true
```
have no effect.

The [doc](https://docs.lvgl.io/8.3/widgets/core/btnmatrix.html#control-buttons) says that for the button matrix `Control buttons` is the way to change states like `disabled`. What we need here is to clear/set the `disabled` flag in the `lvgl.widget.enable` / `lvgl.widget.disable` and `lvgl.widget.show` / `lvgl.widget.hide` shorthands for **all** the buttons with:

> To set/clear the same control attribute for all buttons of a button matrix, use `lv_btnmatrix_set_btn_ctrl_all(btnm, LV_BTNM_CTRL_...`) and `lv_btnmatrix_clear_btn_ctrl_all(btnm, LV_BTNMATRIX_CTRL_...)`.

clydebarrow, 2024-02-06:
To disable an entire button matrix you can now just do this:
```
                - lvgl.widget.update:
                    id: b_matrix
                    state:
                      disabled: true
```

nagyrobi, 2024-02-07:
> To disable an entire button matrix you can now just do this:
> 
> ```
>                 - lvgl.widget.update:
>                     id: b_matrix
>                     state:
>                       disabled: true
> ```

Doesn't work...

clydebarrow, 2024-02-07:
`
            on_press:
              then:
                - lvgl.widget.disable: keypad_matrix
`

clydebarrow, 2024-02-07:
And 
```
        - btn:
            x: 100
            y: 200
            widgets:
              - label:
                  text: Disable keypad
            on_press:
              then:
                - lvgl.widget.update:
                    id: keypad_matrix
                    state:
                      disabled: true
```
Also works.

nagyrobi, 2024-02-07:
It only works in the aspect of not taking interactions, but disabled styles are not applied.