clydebarrow/esphome #58: Background image

nagyrobi, 2024-02-04:
An option to set _display_ (not widget) backdround image/color/opacity:

https://docs.lvgl.io/8/overview/display.html#background

clydebarrow, 2024-02-05:
`disp_bg_color:` is working, `disp_bg_image` is not currently.

clydebarrow, 2024-02-07:
Now working.

closed 2024-02-07 by clydebarrow

nagyrobi, 2024-05-02:
```yaml
color:
  - id: color_pink
    hex: ff00ff

lvgl:
...
  disp_bg_color: color_pink #nor 0xff00ff
```

results in:

```
Compiling .pioenvs/esp32-test-lvgl-host/src/main.o
config/_02_host_lvgl_5_demos.yaml: In lambda function:
config/_02_host_lvgl_5_demos.yaml:2327:37: error: could not convert ‘color_pink’ from ‘esphome::Color’ to ‘lv_color_t’ {aka ‘lv_color16_t’}
*** [.pioenvs/esp32-test-lvgl-host/src/main.o] Error 1
```

nagyrobi, 2024-05-02:
Also, I get a warning
`WARNING The image "config/inc/background1.png" you requested is very big. Please consider using the resize parameter.`
for a 534x534px image.

I think this kind of warning is outdated these days, there are (and will be more and more) displays with resolutions in this range, people would likely want to use wallpapers on.

clydebarrow, 2024-05-07:
> config/_02_host_lvgl_5_demos.yaml:2327:37: error: could not convert ‘color_pink’ from ‘esphome::Color’ to ‘lv_color_t’ {aka ‘lv_color16_t’}
> *** [.pioenvs/esp32-test-lvgl-host/src/main.o] Error 1

Fixed.

clydebarrow, 2024-05-07:
> Imho this kind of warning is outdated these days, there are (and will be more and more) displays with resolutions in this range, people would likely want to use wallpapers on. Possible mitigations:

Make a PR to remove this check, I would suggest.

nagyrobi, 2024-05-07:
OK, I'll make one after LVGL will be merged (so that there will be a reason to do so).