clydebarrow/esphome #62: `obj` width not enforcing

nagyrobi, 2024-02-07:
If i have a child widget wider than the width set, scrollbar will appear at the bottom.
```
        - obj:
            x: 5
            y: 35
            pad_all: 0
            width: 230
            max_width: 230
            height: 250
            bg_opa: 0
            border_width: 1
            widgets:
              - checkbox:
                  id: chk_stat_6
                  text: "Pretty long blah blah blah to go wider than obj width"
                  x: 5
                  y: 5
```
According to LVGL doc

> Children can be visible only within their parent’s bounding area. In other words, the parts of the children outside the parent are clipped.

But this doesn't seem to be respected with `obj`.

clydebarrow, 2024-02-07:
that's the expected behaviour - the child is not visible outside the parent, but the parent becomes scrollable as it can't show all its children at once.

To suppress this use the config option I just added: `scrollbar_mode: "off"` (note - the double quotes are required as YAML treats `off` as meaning `False`.

closed 2024-02-07 by clydebarrow

nagyrobi, 2024-02-08:
> To suppress this use the config option I just added: `scrollbar_mode: "off"` (note - the double quotes are required as YAML treats `off` as meaning `False`.

Only for `obj`? Any other widgets with this?

clydebarrow, 2024-02-08:
All objects. https://docs.lvgl.io/8.3/overview/scroll.html?highlight=scrollbar_mode