clydebarrow/esphome #64: Currently active page variable or sensor

nagyrobi, 2024-02-09:
For uses like:
```yaml
script:
  - id: back_to_main_page_after_delay
    mode: restart
    then:
      - delay: 120s
      - if:
          condition:
            id.of.active page != main_page
          then:
            -  lvgl.page.show: main_page
```

nagyrobi, 2024-02-09:
Oh. just noticed I already mentioned something like this https://github.com/clydebarrow/esphome/issues/9#issuecomment-1887452373

clydebarrow, 2024-02-09:
ESPHome automations don't permit expressions as such (except in lambdas) so it's not possible to have something like "!=" in a condition. Conditions are only boolean values, so something like the existing display condition:
```
  not:
      lvgl.is_displaying_page: main_page
```
would be possible (not implemented yet!)

But this is what I'm using ATM which works just fine.
```
lvgl:
  on_idle:
    timeout: 15s
    then:
      - lvgl.page.show: main_page
  ```

nagyrobi, 2024-02-09:
This comes to my mind, that how many timeouts can we configure?
Is this possible?

```
lvgl:
  on_idle:
  - timeout: 15s
    then:
      - lvgl.page.show: main_page
  - timeout: 60s
    then:
      - light.turn_off: display_backlight
  - ...further timeouts
```

clydebarrow, 2024-02-09:
As many as you want. That was the whole point :-)

nagyrobi, 2024-02-13:
The second timeout doesn't trigger for me:
```
lvgl:
  on_idle:
  - timeout: 15s
    then:
      - light.turn_off: display_backlight
      - lvgl.pause:
  - timeout: 60s
    then:
      - lvgl.page.show: main_page
  - ...further timeouts
```
I assume, the timeout counter doesn't increase while LVGL is paused...?