clydebarrow/esphome #66: `lvgl.is_idle` condition, timeout required

nagyrobi, 2024-02-09:
```yaml
    on_state:
      - if:
          condition: 
            lvgl.is_idle:
          then:
            ...
```
results in
```yaml
  on_state:
    - if:
        condition:

          'timeout' is a required option for [lvgl.is_idle].
          lvgl.is_idle: {}
        then:
```
Is this expected?
All I wanted is to run the automation only when lvgl is in idle state, don't care about timeout here (which should be expired anyway)

nagyrobi, 2024-02-09:
In main lvgl config, I understand. But we are past that. If you think on a timeline, after the timeout specified in the main lvgl config, idle state turns on. Now if I ask lvgl is it idle, how does it matter the duration? it is already in idle state (which is a binary thing as I understand) since who knows when?

nagyrobi, 2024-02-09:
Ok so `timeout` means in reality *for*, like it's been *idle for 20 seconds"

nagyrobi, 2024-02-09:
Please confirm if the statement below is correct:

``lvgl.is_idle``

This condition checks if since the last touch event the amount of time specified here has passed or not.

- **timeout** (**Required**, templatable): Time to check against the time that has elapsed since the last touch event.

Example
```
    # In some trigger:
    on_...:
      then:
        - if:
            condition: lvgl.is_idle
              timeout: 5s
            then:
              - light.turn_off:
                  id: display_backlight
                  transition_length: 3s
```

nagyrobi, 2024-02-09:
Hm...

```yaml
binary_sensor:
  - platform: template
    name: Éjjeli mód
    id: bs_night_mode
    icon: mdi:lightbulb-night-outline
    lambda: |-
        return (!(
          id(cover_ebedlo_pos).state > 20 or
          id(cover_szomszed_pos).state > 20 or
          id(cover_utca_pos).state > 20 or
          id(ts_nappali_4x).state.compare(std::string{"on"}) == 0 or
          id(ts_nappali_4y).state.compare(std::string{"on"}) == 0 or
          id(relay_1).current_values.is_on() or
          id(relay_2).current_values.is_on()
        ));
    on_state:
      - if:
          condition: 
            lvgl.is_idle:
              timeout: 5s
          then:
            if:
              condition:
                - binary_sensor.is_on: ha_connected
              then:
                - script.execute: set_display_idle
```
results in:

```
config/lcd-vezerlo-nappali.yaml: In lambda function:
config/lcd-vezerlo-nappali.yaml:571:33: error: unable to find numeric literal operator 'operator""s'
config/lcd-vezerlo-nappali.yaml: In function 'void setup()':
config/lcd-vezerlo-nappali.yaml:570:45: error: cannot convert 'setup()::<lambda(esphome::lvgl::LvglComponent*)>' to 'std::function<bool(esphome::lvgl::LvglComponent*)>'
In file included from src/esphome.h:66,
                 from src/main.cpp:3:
src/esphome/components/lvgl/lvgl_esphome.h:475:66: note:   initializing argument 1 of 'void esphome::lvgl::LvglCondition<Ts>::set_condition_lambda(std::function<bool(esphome::lvgl::LvglComponent*)>) [with Ts = {bool}]'
  475 |   void set_condition_lambda(std::function<bool(LvglComponent *)> condition_lambda) {
      |                             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
src/esphome/components/lvgl/lvgl_esphome.h: At global scope:
src/esphome/components/lvgl/lvgl_esphome.h:38:19: warning: 'lv_color_t esphome::lvgl::lv_color_from(esphome::Color)' defined but not used [-Wunused-function]
   38 | static lv_color_t lv_color_from(Color color) { return lv_color_make(color.red, color.green, color.blue); }
      |                   ^~~~~~~~~~~~~
*** [.pioenvs/lcd-vezerlo-nappali/src/main.o] Error 1

```

clydebarrow, 2024-02-09:
Fixed. Also noticed that the timeout did not happen until after first touch, have now reset the activity time to zero on startup.

closed 2024-02-09 by clydebarrow