clydebarrow/esphome #94: VNC and `graphical_display_menu` don't play nicely together

nagyrobi, 2024-04-05:
```bash
Failed config

graphical_display_menu: [source config/_02_host_VNC_only_lcdmenu.yaml:73]
  id: my_graphical_display_menu

  ID 'vnc_display' of type vnc::VNCDisplay doesn't inherit from display::DisplayBuffer. Please double check your ID is pointing to the correct value.
  display: vnc_display
  on_redraw:
    - then:
        - component.update:
            id: vnc_display
  active: False
  font: roboto
  mode: rotary
  items:
    - text: Menu Item 1
      type: label
    - text: Menu Item 2
      type: label
```

trying config:

```yaml
substitutions:
  device_name: esp32-test-vnc-host
  friendly_name: "VNC test host"
  device_description: "LVGL host VNC only"

external_components:
  - source: github://clydebarrow/esphome@remote_display
    refresh: 1min
    components: [ vnc ]

esphome:
  friendly_name: "${friendly_name}"
  name: ${device_name}
  comment: "${device_description}"
  platformio_options:
    build_unflags: -Werror=all

host:
  mac_address: "xx:xx:xx:xx:xx:xx"

logger:
  level: DEBUG

api:
  reboot_timeout: 0min

button:
  - platform: restart
    name: ${friendly_name} restart
  - platform: factory_reset
    name: ${friendly_name} factory reset

display:
  - platform: vnc
    id: vnc_display
    update_interval: 1s
    auto_clear_enabled: true
    dimensions:
      width: 720 
      height: 320
    pages:
    - id: graph_page
      lambda: |-
        it.print(0, 0, id(roboto), "My menu is not currently active");

font:
  - file: "gfonts://Roboto"
    id: roboto
    size: 20

graphical_display_menu:
  id: my_graphical_display_menu
  display: vnc_display
  on_redraw:
    then:
      component.update: vnc_display
  active: false
  font: roboto
  mode: rotary
  items:
    - type: Label
      text: Menu Item 1
    - type: Label
      text: Menu Item 2

binary_sensor:
  - platform: status
    name: ${friendly_name} conneted
  - platform: template
    name: "Enter"
    on_press:
      - if:
          condition:
            display_menu.is_active: my_graphical_display_menu
          then:
            - display_menu.enter: my_graphical_display_menu
          else:
            - display_menu.show:  my_graphical_display_menu

  - platform: template
    name: "Menu Up"
    on_press:
      - display_menu.up: my_graphical_display_menu

  - platform: template
    name: "Menu Down"
    on_press:
      - display_menu.down: my_graphical_display_menu

touchscreen:
  - platform: vnc
    id: vnc_touch
    display: vnc_display

sensor:
  - platform: uptime
    name: ${friendly_name} uptime
    disabled_by_default: false
```

nagyrobi, 2024-04-05:
@nielsnl

clydebarrow, 2024-04-23:
I've fixed graphical_display_menu to only require Display, not DisplayBuffer.  The example above builds and runs,
but doesn't display a menu. I have no experience with the menus so that might be right.

Try this:

```
external_components:
  - source: github://clydebarrow/esphome@remote_display
    refresh: 1min
    components: [vnc, graphical_display_menu]
```

nagyrobi, 2024-04-23:
Works - example config above compiles nicely and displays the menu. You can raise the PR for this.

A side question - how could we use antialias on the rendered menu?

closed 2024-04-23 by clydebarrow