clydebarrow/esphome #110: Lvgl on e-paper crash (TTGO T5_V2.3_2.13)

4ib3r, 2024-06-07:
**Describe the bug**
Hello, I try to display simple label on e-paper display, but it crash on start, and don't display anything.

**To Reproduce**
Lvgl component from commit: 8b113e0afd9f8c88e9960189ae7baa970aa9dd81

My config:
```yaml
spi:
  clk_pin: 18
  mosi_pin: 23
  
display:
  - platform: waveshare_epaper
    model: '2.13in-ttgo-b73'
    id: epaper
    cs_pin: 5
    dc_pin: 17
    reset_pin: 16
    busy_pin: 4
    rotation: 0
    full_update_every: 60
    update_interval: never
    # 250x122

lvgl:
  id: mygui
  color_depth: 1
  buffer_size: 25%
  bg_color: 0
  log_level: TRACE
  text_color: 1
  widgets:
    - label:
        id: clock_label
        align: TOP_MID
        text: "00:00:00"  
```
Esphome log:
```

[I][logger:156]: Log initialized
[I][app:029]: Running through setup()...
[I][lvgl:000]: [Info]	(0.259, +227)	 lv_obj_create: begin 	(in lv_obj.c line #215)
[I][lvgl:000]: [Info]	(0.260, +1)	 lv_obj_create: begin 	(in lv_obj.c line #215)
[I][lvgl:000]: [Info]	(0.271, +11)	 lv_obj_create: begin 	(in lv_obj.c line #215)
[I][lvgl:000]: [Info]	(0.282, +11)	 lv_label_create: begin 	(in lv_label.c line #75)
[W][component:157]: Component wifi set Warning flag: unspecified
[I][lvgl:000]: [Info]	(0.402, +120)	 lv_obj_update_layout: Layout update begin 	(in lv_obj_pos.c line #314)
[I][lvgl:000]: [Trace]	(0.415, +13)	 lv_obj_update_layout: Layout update end 	(in lv_obj_pos.c line #317)
[I][lvgl:000]: [Info]	(0.426, +11)	 lv_obj_update_layout: Layout update begin 	(in lv_obj_pos.c line #314)
[I][lvgl:000]: [Trace]	(0.436, +10)	 lv_obj_update_layout: Layout update end 	(in lv_obj_pos.c line #317)
[I][lvgl:000]: [Info]	(0.447, +11)	 lv_obj_update_layout: Layout update begin 	(in lv_obj_pos.c line #314)
[I][lvgl:000]: [Trace]	(0.458, +11)	 lv_obj_update_layout: Layout update end 	(in lv_obj_pos.c line #317)
[I][lvgl:000]: [Info]	(0.468, +10)	 lv_obj_update_layout: Layout update begin 	(in lv_obj_pos.c line #314)
[I][lvgl:000]: [Trace]	(0.478, +10)	 lv_obj_update_layout: Layout update end 	(in lv_obj_pos.c line #317)
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40178319  PS      : 0x00060d30  A0      : 0x800ea6c2  A1      : 0x3ffb2490  
A2      : 0x3ffb4174  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3f403ab7  
A6      : 0x00000001  A7      : 0x3f403a60  A8      : 0x00000000  A9      : 0x3ffb2460  
A10     : 0x00000000  A11     : 0x3ffb487c  A12     : 0x3ffb2488  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x0000003d  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000000c  LBEG    : 0x40089444  LEND    : 0x4008944f  LCOUNT  : 0x00000000  

Backtrace:0x40178316:0x3ffb24900x400ea6bf:0x3ffb24b0 0x400eb2db:0x3ffb2570 0x400efef0:0x3ffb2600 0x400df962:0x3ffb2630 0x40177939:0x3ffb2660 0x401779d5:0x3ffb2680 0x400de352:0x3ffb26a0 0x400e049d:0x3ffb26d0 0x400f94f2:0x3ffb2820 
```
Decoded stack trace:
```
0x40178319: lv_disp_get_draw_buf at /data/build/ogrod-epd/.piolibdeps/ogrod-epd/lvgl/src/hal/lv_hal_disp.c:557
0x40178316: lv_disp_get_draw_buf at /data/build/ogrod-epd/.piolibdeps/ogrod-epd/lvgl/src/hal/lv_hal_disp.c:555
0x400ea6bf: draw_buf_flush at /data/build/ogrod-epd/.piolibdeps/ogrod-epd/lvgl/src/core/lv_refr.c:1269
0x400ea6bf: refr_area_part at /data/build/ogrod-epd/.piolibdeps/ogrod-epd/lvgl/src/core/lv_refr.c:796
0x400eb2db: refr_area at /data/build/ogrod-epd/.piolibdeps/ogrod-epd/lvgl/src/core/lv_refr.c:678
0x400eb2db: refr_invalid_areas at /data/build/ogrod-epd/.piolibdeps/ogrod-epd/lvgl/src/core/lv_refr.c:616
0x400eb2db: _lv_disp_refr_timer at /data/build/ogrod-epd/.piolibdeps/ogrod-epd/lvgl/src/core/lv_refr.c:324
0x400efef0: lv_timer_exec at /data/build/ogrod-epd/.piolibdeps/ogrod-epd/lvgl/src/misc/lv_timer.c:313
0x400efef0: lv_timer_handler at /data/build/ogrod-epd/.piolibdeps/ogrod-epd/lvgl/src/misc/lv_timer.c:109
0x400df962: lv_timer_handler_run_in_period at /data/build/ogrod-epd/.piolibdeps/ogrod-epd/lvgl/src/misc/lv_timer.h:85
0x400df962: esphome::lvgl::LvglComponent::loop() at /data/build/ogrod-epd/src/esphome/components/lvgl/lvgl_esphome.h:383
0x40177939: esphome::Component::call_loop() at /data/build/ogrod-epd/src/esphome/core/component.cpp:77
0x401779d5: esphome::Component::call() at /data/build/ogrod-epd/src/esphome/core/component.cpp:104
0x400de352: esphome::Application::setup() at /data/build/ogrod-epd/src/esphome/core/application.cpp:52
0x400e049d: setup() at /data/build/ogrod-epd/src/main.cpp:345
0x400f94f2: loopTask(void*) at /data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42
```
This board doesn't has psram and this causing crash when lvgl buffer is 100%. If I think correctly with smaller buffer it should work.
Anyone have any ideas how to solve this problem?

clydebarrow, 2024-06-07:
1 bit color mode is not supported. Use the most recent code.

closed 2024-06-07 by clydebarrow

clydebarrow, 2024-06-07:
And just FYI, I have tested this on an Inkplate 6 Plus in monochrome mode (but using 16 bit color depth in LVGL) and it works ok. Slow, obviously, and you need to make all backgrounds pure black, but it does work. Better support for monochrome and e-ink will come in the future.