clydebarrow/esphome #111: Compile error in LVGL after last commit.

AnHardt, 2024-06-13:
**Describe the bug**
After the last lvgl commit '12.6.2024' i get.
```
src/esphome/components/lvgl/lvgl_esphome.cpp: In function 'void* lv_custom_mem_alloc(size_t)':
src/esphome/components/lvgl/lvgl_esphome.cpp:38:61: error: expected unqualified-id before ';' token
   esphome::esph_log_v(TAG, "allocate %zu - > %p", size, ptr);
                                                             ^
src/esphome/components/lvgl/lvgl_esphome.cpp: In function 'void lv_custom_mem_free(void*)':
src/esphome/components/lvgl/lvgl_esphome.cpp:43:43: error: expected unqualified-id before ';' token
   esphome::esph_log_v(TAG, "free %p", ptr);
                                           ^
src/esphome/components/lvgl/lvgl_esphome.cpp: In function 'void* lv_custom_mem_realloc(void*, size_t)':
src/esphome/components/lvgl/lvgl_esphome.cpp:50:57: error: expected unqualified-id before ';' token
   esphome::esph_log_v(TAG, "realloc %p: %zu", ptr, size);
                                                         ^
*** [.pioenvs/az2/src/esphome/components/lvgl/lvgl_esphome.cpp.o] Error 1
```

clydebarrow, 2024-06-13:
Can't reproduce that. Yaml?

AnHardt, 2024-06-13:
```jaml
substitutions:
  name: lvgl-test
  friendly_name: LVGL Test
  device_description: "LVGL Display in Experimental"

  screen_backlight_pwm_pin: '15'
  tft_spi_clk_pin: GPIO18
  tft_spi_mosi_pin: GPIO23
  tft_spi_miso_pin: GPIO19
  tft_cs_pin: GPIO5
  tft_dc_pin: GPIO4
  tft_reset_pin: GPIO22
  touchscreen_cs_pin: GPIO14
  touchscreen_interrupt_pin: GPIO27

api:
  encryption:
    key: "usPaSMTPYs3PSxP5T4vdm2cvuebApj81wO1QNdK/6sx="

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  comment: ${device_description}

external_components:
  - source: github://pr#6363
    components: [lvgl]
    refresh: 1min

esp32:
  board: esp32dev
  framework:
    type: arduino

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  id: internal_wifi

output:
  - platform: ledc
    pin: ${screen_backlight_pwm_pin}
    id: screen_backlight_pwm
    inverted: True
    channel: 1

light:
  - platform: monochromatic
    output: screen_backlight_pwm
    name: "Hintergrundbeleuchtung"
    restore_mode: RESTORE_DEFAULT_ON
    id: local_light

spi:
  - id: tft
    clk_pin: ${tft_spi_clk_pin}
    mosi_pin: ${tft_spi_mosi_pin}
    miso_pin: ${tft_spi_miso_pin}

touchscreen:
  platform: xpt2046
  id: my_touch
  spi_id: tft
  cs_pin: ${touchscreen_cs_pin}
  interrupt_pin: ${touchscreen_interrupt_pin}
  #update_interval: 50ms
  threshold: 400
  calibration:
    x_min: 300
    x_max: 3850
    y_min: 360
    y_max: 3890
  transform:
    mirror_x: true
    mirror_y: false
    swap_xy: false
  on_touch:
    - delay: 50ms
    - light.turn_on: local_light
    - lambda: |-
          ESP_LOGI("cal", "x=%d, y=%d, x_raw=%d, y_raw=%0d",
              touch.x,
              touch.y,
              touch.x_raw,
              touch.y_raw
              );              

display:
  - platform: ili9xxx
    model: ili9341
    spi_id: tft
    cs_pin: ${tft_cs_pin}
    dc_pin: ${tft_dc_pin}
    reset_pin: ${tft_reset_pin} 
    transform:
      swap_xy: false
      mirror_x: false
      mirror_y: true
    dimensions:
      height: 320
      width: 240
    auto_clear_enabled: false
    update_interval: never
    id: screen

lvgl:
  displays:
    - screen
  touchscreens:
    - my_touch
  log_level: NONE
  buffer_size: 25%
  pages:
    - id: main_page
      widgets:
        - label:
            align: CENTER
            text: 'Hello World!'
```
```
INFO ESPHome 2024.5.5
INFO Reading configuration /config/esphome/lvgl-test.yaml...
WARNING GPIO15 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing lvgl-test (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.4 has been installed!
INFO noise-c@0.1.4 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.1
INFO Installing esphome/libsodium @ 1.10018.1
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.1 has been installed!
INFO libsodium@1.10018.1 has been installed!
Library Manager: Installing lvgl/lvgl @ 8.4.0
INFO Installing lvgl/lvgl @ 8.4.0
Unpacking  [####################################]  100%
Library Manager: lvgl@8.4.0 has been installed!
INFO lvgl@8.4.0 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|-- SPI @ 2.0.0
|-- lvgl @ 8.4.0
Compiling .pioenvs/lvgl-test/src/esphome/components/api/api_connection.cpp.o
...
Compiling .pioenvs/lvgl-test/src/esphome/components/lvgl/lvgl_esphome.cpp.o
Compiling .pioenvs/lvgl-test/src/esphome/components/mdns/mdns_component.cpp.o
src/esphome/components/lvgl/lvgl_esphome.cpp: In function 'void* lv_custom_mem_alloc(size_t)':
src/esphome/components/lvgl/lvgl_esphome.cpp:35:67: error: expected unqualified-id before ';' token
     esphome::esph_log_e(TAG, "Failed to allocate %zu bytes", size);
                                                                   ^
src/esphome/components/lvgl/lvgl_esphome.cpp:38:61: error: expected unqualified-id before ';' token
   esphome::esph_log_v(TAG, "allocate %zu - > %p", size, ptr);
                                                             ^
src/esphome/components/lvgl/lvgl_esphome.cpp: In function 'void lv_custom_mem_free(void*)':
src/esphome/components/lvgl/lvgl_esphome.cpp:43:43: error: expected unqualified-id before ';' token
   esphome::esph_log_v(TAG, "free %p", ptr);
                                           ^
src/esphome/components/lvgl/lvgl_esphome.cpp: In function 'void* lv_custom_mem_realloc(void*, size_t)':
src/esphome/components/lvgl/lvgl_esphome.cpp:50:57: error: expected unqualified-id before ';' token
   esphome::esph_log_v(TAG, "realloc %p: %zu", ptr, size);
                                                         ^
Compiling .pioenvs/lvgl-test/src/esphome/components/mdns/mdns_esp32.cpp.o
*** [.pioenvs/lvgl-test/src/esphome/components/lvgl/lvgl_esphome.cpp.o] Error 1
========================= [FAILED] Took 18.76 seconds =========================
```

nagyrobi, 2024-06-13:
Getting similar:
```
src/esphome/components/lvgl/lvgl_esphome.cpp: In function 'void* lv_custom_mem_alloc(size_t)':
src/esphome/components/lvgl/lvgl_esphome.cpp:38:61: error: expected unqualified-id before ';' token
   esphome::esph_log_v(TAG, "allocate %zu - > %p", size, ptr);
                                                             ^
src/esphome/components/lvgl/lvgl_esphome.cpp: In function 'void lv_custom_mem_free(void*)':
src/esphome/components/lvgl/lvgl_esphome.cpp:43:43: error: expected unqualified-id before ';' token
   esphome::esph_log_v(TAG, "free %p", ptr);
                                           ^
src/esphome/components/lvgl/lvgl_esphome.cpp: In function 'void* lv_custom_mem_realloc(void*, size_t)':
src/esphome/components/lvgl/lvgl_esphome.cpp:50:57: error: expected unqualified-id before ';' token
   esphome::esph_log_v(TAG, "realloc %p: %zu", ptr, size);
                                                         ^
*** [.pioenvs/lcd-kezelo-nappali/src/esphome/components/lvgl/lvgl_esphome.o] Error 1
```

clydebarrow, 2024-06-13:
Ah, it fails with no `logger:`.

clydebarrow, 2024-06-13:
I reverted the last commit. Will revisit it in due course.

closed 2024-06-13 by clydebarrow

AnHardt, 2024-06-14:
Works again.