clydebarrow/esphome #117: FR: `udp` multiple ports

nagyrobi, 2024-06-17:
This may be a bit too much, I admit...

Support for separation of provided data to different IPs. Like:
```yaml
udp:
  - id: udp_internal
    update_interval: 5s
    encryption: "REPLACEME A"
    port: 18511
    addresses:
      - 255.255.255.255
    sensors:
      - temp_rooma
      - temp_roomb
      - temp_roomc
      - temp_floor1
      - temp_hall1
      - temp_hall2
      - temp_floor2
      - temp_water1
      - temp_water2
      - temp_garage
      - temp_outdoor
      - humi_rooma
      - humi_roomb
      - humi_roomc

  - id: udp_external
    update_interval: 5s
    encryption: "REPLACEME B"
    port: 18512
    addresses:
      - 10.87.135.110
    binary_sensors:
      - binary_sensor_id1
    sensors:
      - temp_outdoor
```
Some use cases:
- one wants to provide a subset of sensors to a receiver in another network
- one wants to provide different sets of sensors to different IPs and/or ports

clydebarrow, 2024-06-18:
Should work now.

closed 2024-06-18 by clydebarrow

nagyrobi, 2024-06-20:
Doc updated with an example showing this.