clydebarrow/esphome #119: Additional ST7701s initialisation sequences

sirriseyrek, 2024-07-11:
Any solution to add for ST7701S Lily Go T-RGB round display initialization code. Original and running code available on [https://github.com/Xinyuan-LilyGO/LilyGo-T-RGB/tree/main/src](url) but the code on ESP home not matching with the display. Init part from code below

`DRAM_ATTR static const lcd_init_cmd_t st7701_2_1_inches[] = {
    {0xFF, {0x77, 0x01, 0x00, 0x00, 0x10}, 0x05},
    {0xC0, {0x3b, 0x00}, 0x02},
    {0xC1, {0x0b, 0x02}, 0x02},
    {0xC2, {0x07, 0x02}, 0x02},
    {0xCC, {0x10}, 0x01},
    {0xCD, {0x08}, 0x01}, 
    {0xb0, {0x00, 0x11, 0x16, 0x0e, 0x11, 0x06, 0x05, 0x09, 0x08, 0x21, 0x06, 0x13, 0x10, 0x29, 0x31, 0x18}, 0x10},
    {0xb1, {0x00, 0x11, 0x16, 0x0e, 0x11, 0x07, 0x05, 0x09, 0x09, 0x21, 0x05, 0x13, 0x11, 0x2a, 0x31, 0x18}, 0x10},
    {0xFF, {0x77, 0x01, 0x00, 0x00, 0x11}, 0x05},
    {0xb0, {0x6d}, 0x01},
    {0xb1, {0x37}, 0x01},
    {0xb2, {0x81}, 0x01},
    {0xb3, {0x80}, 0x01},
    {0xb5, {0x43}, 0x01},
    {0xb7, {0x85}, 0x01},
    {0xb8, {0x20}, 0x01},
    {0xc1, {0x78}, 0x01},
    {0xc2, {0x78}, 0x01},
    {0xc3, {0x8c}, 0x01},
    {0xd0, {0x88}, 0x01},
    {0xe0, {0x00, 0x00, 0x02}, 0x03},
    {0xe1, {0x03, 0xa0, 0x00, 0x00, 0x04, 0xa0, 0x00, 0x00, 0x00, 0x20, 0x20}, 0x0b},
    {0xe2, {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00}, 0x0d},
    {0xe3, {0x00, 0x00, 0x11, 0x00}, 0x04},
    {0xe4, {0x22, 0x00}, 0x02},
    {0xe5, {0x05, 0xec, 0xa0, 0xa0, 0x07, 0xee, 0xa0, 0xa0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00}, 0x10},
    {0xe6, {0x00, 0x00, 0x11, 0x00}, 0x04},
    {0xe7, {0x22, 0x00}, 0x02},
    {0xe8, {0x06, 0xed, 0xa0, 0xa0, 0x08, 0xef, 0xa0, 0xa0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00}, 0x10},
    {0xeb, {0x00, 0x00, 0x40, 0x40, 0x00, 0x00, 0x00}, 0x07},
    {0xed, {0xff, 0xff, 0xff, 0xba, 0x0a, 0xbf, 0x45, 0xff, 0xff, 0x54, 0xfb, 0xa0, 0xab, 0xff, 0xff, 0xff}, 0x10},
    {0xef, {0x10, 0x0d, 0x04, 0x08, 0x3f, 0x1f}, 0x06},
    {0xFF, {0x77, 0x01, 0x00, 0x00, 0x13}, 0x05},
    {0xef, {0x08}, 0x01},
    {0xFF, {0x77, 0x01, 0x00, 0x00, 0x00}, 0x05},
    {0x36, {0x08}, 0x01},       
    {0x3a, {0x66}, 0x01}, 
    {0x11, {0x00}, 0x80},
    {0x29, {0x00}, 0x80},
    {0, {0}, 0xff}
};`

clydebarrow, 2024-07-11:
I would suggest comparing the two sets of init sequences and either replacing, or appending to, the default ones with any others required (in YAML.)

sirriseyrek, 2024-07-16:
Thank you @clydebarrow  I already recorded original init over (LilyGo-T-Rgb 2.1) SPI pins with logic analyzer. But on Yaml "init_sequence:" is waiting for us to choose one of the defined canned sequences. I think there are differences between round display and square display. What should I do to add the init code that has differences instead of the defined (-1) init sequence? If I don't put -1 under "init_sequence:" it gives an error.

clydebarrow, 2024-07-16:
> If I don't put -1 under "init_sequence:" it gives an error.

If you have a problem with it, attach the YAML and the error output. Just saying "it gives an error" is not useful if you don't say what "it" is or what the error said. The standard sequence ("1" not "-1"; the "-" is just the YAML array marker) is not required if you provide your own sequence.

sirriseyrek, 2024-07-16:
Thank you for the information. Got it  - 1 is array marker. FYI All of my files are below

[esphome-web-cf1ed4.txt](https://github.com/user-attachments/files/16241976/esphome-web-cf1ed4.txt)
Here is the Yaml file I am using. also added excel file  
[Factory -EspHome compare.xlsx](https://github.com/user-attachments/files/16242024/Factory.-EspHome.compare.xlsx)
it is compare between the LilyGo - EspHome inits. Trouble is not error no any error on log but also no any sign of text on the screen with esphome setup I'm adding log too
[logs_esphome-web-cf1ed4_logs (1).txt](https://github.com/user-attachments/files/16242018/logs_esphome-web-cf1ed4_logs.1.txt)

clydebarrow, 2024-07-16:
Your yaml just has this:
```
    init_sequence:
      - [0x00]
      - 1 # select canned init sequence number 1
     # - [ 0xE0, 0x1F ]  # Set sunlight readable enhancement
     # - [0xFF, 0x77, 0x01, 0x00, 0x00, 0x10] # CMD2_BKSEL_BK0
     # - [ 0xCD, 0x00 ]
```
All that does is send a 0 command before the canned sequence. The capture file is no help because it doesn't distinguish command from data bytes, nor does it bear any obvious relation to the init sequence anyway.

Are you using the 2.1" or the 2.8"? The 2.1" init sequence in Lilygo's source code is identical to the ESPHome version, the 2.8" version is different.

sirriseyrek, 2024-07-16:
Yes I am using Semi circle 2.1 t-rgb. I tested many variations on Yaml.  The Yaml file i send one of the tested actually latest one  and I added 0x00 before canned sequence because even both  firmware seems mode 3 as spi timings  during capture the beginning on logic analyzer starts as high on ESPhome code but all starts with low with lilygo firmware. Just one of the unsuccesful attempt. Also I am  using software spi because the LCD spi pins not connected ESP gpio directly on this board all three spi passing over i2c gpio extender. Maybe ESPhome code not compatible because of the gpio extender additional needs. I'm totally open to all possible ideas. Thanks anyway i will try to dig more.

clydebarrow, 2024-07-16:
> I am using Semi circle 2.1 t-rgb

There is no semi-circle 2.1 listed on the LilyGo site - just 2.1 full circle, 2.8 full circle and 2.8 half circle.