clydebarrow/esphome #120: Import of SVG files not working correctly with LVGL

agillis, 2024-07-19:
**Describe the bug**
If you use and SVG with the image: command and set use_transparency: true most all anti-aliasing is removed from the image. I'm not sure if this is just because the LVGL code only supports RGB565 or another issue.

**To Reproduce**
Steps to reproduce the behavior:
use and SVG with use_transparency: true
This is my test

```
image:
  - file: https://esphome.io/_static/logo-text-on-light.svg
    id: boot_logo
    type: RGB565
    resize: 470x95
    use_transparency: true
```    

**Expected behavior**
Image should be clear on the background with only the transparent part replaced with the background color

**Screenshots**
See my bug https://github.com/esphome/issues/issues/6047

**Desktop (please complete the following information):**

**Smartphone (please complete the following information):**

**Additional context**

clydebarrow, 2024-07-19:
Anti-aliasing with transparency isn't possible with RGB565 since there is no alpha channel.

closed 2024-07-19 by clydebarrow

agillis, 2024-07-19:
Would it be possible to support  image type RGBA in the LVGL code? Otherwise it's not possible to use SVG files and have them look good so this feature is essentially useless.

clydebarrow, 2024-07-19:
It should be possible, right now it doesn't work. That will fix the anti-aliasing when it works.

agillis, 2024-07-19:
OK thanks. I have done a lot of testing of the LVGL code over the past moth or so and this is the only outstanding problem I have seen. Thanks for all your hard work!